frankarobotics/libfranka #151: Controlling franka gripper in real time

sdfzz, 2024-02-28:
Hi all,

I'm trying to use a franka research 3 robot as a follower robot of a teleoperation system. 

I want the operator to be able to freely open/close the gripper and grasp objects with various shapes; thus, I figured that "grasp()" function provided by libfranka may not be the best choice. 

So, I tried to use "move()" function instead, but got following problems:

1. using "move()" function inside realtime loop caused "communication_constraints_violation" error
2. when I placed "move()" in other loop, I can control the gripper, but it seems that the function freezes when the gripper grasped the object (ie. it cannot move to the desired position). Upon grasping an object, the gripper control loop freezed.

What is the best way to control the franka gripper in realtime to grasp and move objects with various size without ROS?

Any help would be appreciated

Regards,

Steve

Silverwings-zero, 2025-04-09:
Hi,

I was wondering if you’ve managed to solve the issue, as I’ve encountered a similar problem. I also need real-time control of the Panda gripper, including force control, but I haven’t been able to find any resources online that support this.

Best regards,
Michael

Zwhy2025, 2025-08-10:
@winkma I'm really looking forward to this feature! Real-time gripper control would be a huge help for our research. Thank you for your hard work!