frankarobotics/libfranka #152: Cannot connect to the desk

xzhuzhu, 2024-03-08:
Dear Franka Development Team,

I have been using a Franka robot for three years without any issues. However, recently I have been unable to establish a connection between the robot and the desk. When I attempt to ping the IP address 172.16.0.2, I receive a "destination host unreachable" error message. I have already tried unplugging and replugging all the cables multiple times, but the problem persists. The local IP address assigned to the robot is 172.16.0.1. Additionally, I have replaced the cable connecting the computer and the controller with a new one, but the issue remains unresolved. I would greatly appreciate any assistance or suggestions to resolve this problem. Thank you very much~

JeffWang-zxw20, 2024-08-20:
Hi, I just encountered the same problem. Have you solved it?

AndreasKuhner, 2024-10-09:
Hi @xzhuzhu , @JeffWang-zxw20 and @hussainzeb ,
did you guys check to attach a network cable directly to the arm base from your notebook, activate the DHCP to get automatically an IP from the arm and go to the page _robot.franka.de_? This should give you access to Desk -> maybe something changed with the network settings... :thinking: 

You can also read about this within the user manual at https://franka.de/documents in chapter **11.3 - Initial Configuration**

Cheers,
Andreas

hussainzeb, 2024-10-09:
Hi Andreas @AndreasKuhner ,

Yes, I tried that already but still cannot access it. The robot has stable yellow lights and the ethernet connect directly to it and my laptop. Am i missing something else here?

JeffWang-zxw20, 2024-10-09:
I solved the issue by booting the robot into rescue mode (franka Emika manual page 173) and then reboot from the menu resolved the problem.

hussainzeb, 2024-10-09:
Finally I was able to access the Franka Desk Page but it now asks for a username and a password which I do not remember. @AndreasKuhner  is there a way around that as well? many thanks for your help

AndreasKuhner, 2024-10-09:
@hussainzeb , :sweat_smile: I fear we don't have any username/password recovery I know of - that you need to remember yourself :wink: 

I guess your best take on it would be, as @JeffWang-zxw20 described, to go into the rescue mode and trigger a factory reset. Then you can re-do users (pointing to the FR3 manual https://franka.de/documents)

hussainzeb, 2024-10-09:
@AndreasKuhner  I have an old Franka Emika Panda from 2019. 
I tried the  factory reset as explained in the following link, 

https://www.franka-community.de/t/reset-admin-password-or-reset-to-factory-settings/184

but I am getting syntax error.... any further advice is highly appreicated.

AndreasKuhner, 2024-10-09:
Uhh, Panda :sweat_smile: 

Maybe you can try to follow what calebescobedo suggested to get into the rescue mode. From there you should be able to trigger a factory reset.

hussainzeb, 2024-10-09:
fetch("/admin/api/factory-reset",
{method: "POST"}
)

This command in console of the browser did the trick for me. POST had to be in double quotes, not single as was said in link above. I am not JS expert so don't know the reason but it worked. 

Anyway thanks a lot for all your comments ...

AndreasKuhner, 2025-03-05:
If there is still an open issue, please let us know, thanks all! 😄

closed 2025-03-05 by AndreasKuhner