frankarobotics/libfranka #154: libfranka: Connection to FCI refused. Please install FCI feature or enable FCI mode in Desk

upc-ghy, 2024-03-29:
PC：
Ubuntu 20.04 
Ros noetic
Rt kernel:  Linux xxxxxx 5.15.76-rt53 #2 SMP PREEMPT_RT xxxxxxxxx x86_64 GNU/Linux

I followed the official tutorial at https://frankaemika.github.io/docs/installation_linux.html#installing-from-the-ros-repositories to install libfranka and franka_ros from source code. I also correctly installed the real-time kernel according to the tutorial. However, when I tried to test if the installations were successful, I encountered the following issue.

My PC's browser can successfully connect to Franka Desk, and in the Desk interface, I unlocked the robot, changing its indicator light from yellow to blue. Additionally, when I ping the Franka robot's IP address 172.16.3.69 in the terminal on my PC, I can reach it.

Nevertheless, when I entered the libfranka/build/examples/ directory and ran the test command below, I received the following error. I am unsure where the problem lies and would appreciate your guidance.

```
./communication_test 172.16.3.69
libfranka: Connection to FCI refused. Please install FCI feature or enable FCI mode in Desk.
```

closed 2024-03-30 by upc-ghy

mehranraisi, 2024-12-20:
Hello,

May I ask you how did you solve this problem, as I encounter the same problem, too. I tried different versions, but still get the same problem.

Bests,
Mehran

chrksa, 2025-11-04:
@mehranraisi Hi,
Still have the problem :(. The "FCI is active tab" is open and when i ping the controller, i get an answer, but when executing echo_robot_state i still get the connection refused.
./examples/echo_robot_state 192.168.1.11
libfranka: Connection to FCI refused. Please install FCI feature or enable FCI mode in Desk.

abaicoming, 2025-11-04:
@mehranraisi 

Thank you very much for your opinion.

Actually, I have activated FCI in Desk, and I noticed that Franka's light turned **green**. What's more interesting is that I can read all the states of Franka, including joint positions, velocities, torques, and so on, using the command `./build/examples/echo_robot_state <fci-ip>`. However, when I tried to run the command `roslaunch franka_visualization franka_visualization.launch robot_ip:=172.16.0.2 load_gripper:=true robot:=fr3`, it reported an error: `[ERROR] [1760095263.958655803]: Exception: libfranka: Connection to FCI refused. Please install FCI feature or enable FCI mode in Desk.`

I'll try more later.