frankarobotics/libfranka #157: FR3 - print_joint_poses - Illegal instruction error

eneserciyes, 2024-06-28:
Hello,

We are setting up our FR3s.

They have system version 5.6 and I am using libfranka version 0.13.3. I can compile the library and run all examples that move the robot successfully. However, the example "print_joint_poses" fails with "Illegal instruction error (core dumped)". Here are the screenshots of failure and backtrace from gdb:

[image]

I was able to pinpoint the line of failure:

https://github.com/frankaemika/libfranka/blob/8f52b647bfb5989aec1a023167627431e2db09c0/src/model.cpp#L56

There is a problem with the ModelLibrary class and Poco::SharedLibrary object there. I tried with different Poco versions but I still get the same issues.

Do you have any idea what might cause this problem?

derekgeng15, 2024-06-28:
I have the same error as well, pinpointed the same error line as well

eneserciyes, 2024-06-29:
Robot sends a shared library over network to the controlling device and the error comes from executing a function in that shared library. We realized it had something to do with the specific NUC and its CPU architecture. Trying a different machine (a different NUC version) worked for us.

closed 2024-06-29 by eneserciyes