frankarobotics/libfranka #160: Server Library Compatibility Issue

schromya, 2024-08-20:
ROS 1 version: Noetic
Ubuntu version: 20.04.6
Robot system version: 4.2.2 (pandas)
Robot / Gripper Server version: 5/3
libfranka version: 0.9.2
franka_ros version: 0.9.1

When I run the communication test (`sudo communication_test 192.168.1.2`), I receive the following error:
```bash
libfranka: Incompatible library version (server version: 5, library version: 4). 
Please check https://frankaemika.github.io for Panda system updates or use a different version of libfranka.
```

According to the [compatibility docs](https://frankaemika.github.io/docs/compatibility.html), my installed versions of the libraries should be correct, so I am not sure what the issue is. Any ideas?

Fatma-Dhaoui, 2024-09-06:
Hi,
Please update sub-modules and try again.
git submodule update --init --recursive

Thanks!

schromya, 2024-09-06:
I ran `git submodule update --init --recursive` within  libfranka and recompiled the library. I'm still running into the same issue. 

However, I have discovered that I can run the examples within libfranka (like libfranka/examples/communication_test.cpp).  The incompatibility problem just occurs when I try to use `communication_test` with  franka_ros.

schromya, 2024-09-09:
### Solution(s)
I fixed the issue. It turns out there were a couple problems going on:
1. I had the ros-noetic-libfranka library installed on my machine which was interfering which my version of libfranka built from source like in [this other issue](https://github.com/frankaemika/franka_ros/issues/310). 

    SOLUTION: I uninstalled the library using `sudo apt remove ros-noetic-libfranka` and recompiled my catkin_ws.

2. I didn't build the libfranka library into a debian package since that step was marked as optional in the [installation guide](https://frankaemika.github.io/docs/installation_linux.html). This caused franka_ros to try to use  libfranka 0.8.0 (which I didn't install) instead of the correct 0.9.2 version when I ran `catkin_make -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=../libfranka/build`. [This other issue](https://github.com/frankaemika/franka_ros/issues/257) was similar to the problem I was facing. 

    SOLUTION: I built the debian package using the following commands in my `libfranka/build` directory:
    ```bash
    cpack -G DEB
    sudo dpkg -i libfranka-0.9.2-x86_64.deb
    ```
    Then in my catkin_ws, I could compile franka_ros properly using:
    ```bash
    catkin_make -DCMAKE_BUILD_TYPE=Release 
    source devel/setup.sh
    ```

### Documentation
I ended up running everything in a Docker Container. The directions and Dockerfile for setting up this process are [here](https://github.com/schromya/panda/tree/main/ros_noetic).

closed 2024-09-09 by schromya