frankarobotics/libfranka #163: libfranka:Move command abord motion aborted by reflex ["communication_constraints_violation"]

git183nuozhe, 2024-10-08:
hi, nice see you.
when i use libfranka 0.9.1 "communication_test"only, it is ok. but when run the py script as follows at the same time, the error: terminate called after throwing an instance of 'franka::ControlException' what(): libfranka:Move command aborted:motion aborted by reflex!["communication_constraints_violation"] is comming up. it is not cpu hz problem. what should i do?

the hardware situationa： master computer 192.168.1.10, a franka arm192.168.1.6, a realsense slaver computer192.168.1.13，which all in switch network.

py script 
the py script is running at realsense slaver computer.the  "communication_test"is running at master computer .

#!/usr/bin/env python3
import rospy
import cv2, re, glob
from sensor_msgs.msg import Image
#from sensor_msgs.msg import JointState
from cv_bridge import CvBridge
#import os
from pathlib import Path
#import time

bridge = CvBridge()
frame_count = 0
video_writer = None
video_file = None
is_recording = False

save_dir = None
video_savefile = None

flag_num_q = 1

def increment_path(path, exist_ok=False, sep='', mkdir=False):
    # Increment file or directory path, i.e. save_dir/RGB_video/video, video2, video3, ... etc.
    path = Path(path)  # os-agnostic
    n2 = ''
    if path.exists() and not exist_ok:
        suffix = path.suffix
        path = path.with_suffix('')
        dirs = glob.glob(f"{path}{sep}*")  # similar paths
        matches = [re.search(rf"%s{sep}(\d+)" % path.stem, d) for d in dirs]
        i = [int(m.groups()[0]) for m in matches if m]  # indices
        n2 = max(i) + 1 if i else 2  # increment number
        path = Path(f"{path}{sep}{n2}{suffix}")  # update path
    dir = path if path.suffix == '' else path.parent  # directory
    if not dir.exists() and mkdir:
        dir.mkdir(parents=True, exist_ok=True)  # make directory
    if len(str(n2)) == 0:
        n2 = 1
    return path, str(n2)

def image_callback(msg):
    global frame_count, video_writer, video_file, is_recording, video_savefile, flag_num_q

    try:
        cv_image = bridge.imgmsg_to_cv2(msg, desired_encoding="bgr8")
    except Exception as e:
        rospy.logerr(f"Image conversion failed: {e}")
        return

    ros_time = msg.header.stamp
    time_text = f"ROS Time: {ros_time.secs}.{ros_time.nsecs:09d}"

    font = cv2.FONT_HERSHEY_SIMPLEX
    font_scale = 1
    font_color = (255, 255, 255)  
    thickness = 2
    position = (10, 30)  
    cv_image = cv2.putText(cv_image, time_text, position, font, font_scale, font_color, thickness, cv2.LINE_AA)

    # cv2.imshow('Frame', cv_image)
    #key = cv2.waitKey(1) & 0xFF  
    key_file = Path("key.txt")
    if key_file.exists():
        with open(key_file,'r') as f:
            key = f.read().strip()
    else:
        #print(key)
        rospy.logerr("Error:No key data")
    print(key)
    #if key != 's':
        #if key != 'q':
            #if key != 'p':
                #if key != 'a':
                    #print(key)
                    #rospy.loginfo("Alert:key value is not right")
    if key == 's':
        #rospy.loginfo(f"Now time111: {time.time()}\n")
        if not is_recording:
            is_recording = True
            rospy.loginfo('Recording started...')

            video_dirname, order_name = 'save_dir/RGB_video', '' 
            exist_ok = False
            save_dir, order_num = increment_path(Path(video_dirname) / order_name, exist_ok=exist_ok)
            save_dir = str(save_dir)
            print(order_num)
        
            order_num = int(order_num) if order_num.isnumeric() else ''
            Path(save_dir).mkdir(parents=True, exist_ok=True)

            for char in save_dir:
                if char in str(order_num):
                    save_dir=save_dir.replace(char,'')
            #print(save_dir)

            video_savefile = Path(save_dir) / f'video_{order_num}.mp4c'
            #joint_rosdata_file = Path(save_dir) / f'joint_rosdata_{order_num}.txt'

            height, width, _ = cv_image.shape
            print(height)
            print(width)
            rospy.loginfo(f"Image dimensions: {width}x{height}")
            fourcc = cv2.VideoWriter_fourcc('M', 'J', 'P', 'G')
            #fourcc = cv2.VideoWriter_fourcc(*'mp4v')  
            video_writer = cv2.VideoWriter(str(video_savefile), fourcc, 30, (width, height))
        else:
            rospy.loginfo('Already Recording...')
    if key == 'q':
        if flag_num_q == 0:
            flag_num_q = 1
            print('...video data logging done...')
            print(f'，if save video，the data in：{str(video_savefile)}')
            frame_count = 0

            is_recording = False

    if key == 'p':
        if is_recording and video_writer:
            video_writer.release()
        rospy.loginfo('Recording stopped and exiting...')
        cv2.destroyAllWindows()
        rospy.signal_shutdown('Video saved and exiting...')

    if is_recording:
        #rospy.loginfo(f"Now time222: {time.time()}\n")
        if video_savefile is not None:

            
            video_writer.write(cv_image)
            # xieru joint data
            #record_ros_data(str(joint_rosdata_file))

            frame_count += 1
            rospy.loginfo(f"Frame {frame_count} with timestamp saved to {video_savefile}")
            flag_num_q = 0
        else:
            rospy.loginfo("Recording was not started. video_savefile is not None.")

def main():
    rospy.init_node('realsense_image_saver_with_timestamp', anonymous=True)
    rospy.Subscriber("/camera/color/image_raw", Image, image_callback, queue_size=1)

    try:
        rospy.spin()
    except KeyboardInterrupt:
        rospy.loginfo("Shutting down...")
    finally:
        if video_writer:
            video_writer.release()
            rospy.loginfo("Video writer released.")
        cv2.destroyAllWindows()
if __name__ == '__main__':
    main()

git183nuozhe, 2024-10-08:
the py script is used to read letters from a txt file and determine whether to save the realsense image data in the ROS topic as a video based on the letters

AndreasKuhner, 2024-10-09:
Hi @git183nuozhe ,
it sounds a bit like that your image processing is blocking the communication with the robot. One iteration for the FCI cycle can only take around 500 us (aka receiving data via libfranka, processing it and sending a new command). If something else blocks or delays that, you will see the communcation violation error. 

Maybe you can make sure that the image processing is not blocking/delaying or using the same resource than your communication with the robot via libfranka :smiley: 

Cheers,
Andreas

AndreasKuhner, 2024-11-08:
Hi @git183nuozhe ,
did you find a solution?

mrceki, 2025-03-31:
Hi, I am also having the same issue. But I am not even processing images, e.g. I am just visualizing the external wrenches with plotjuggler and randomly it throws this error, whats your suggestion on that?

AndreasKuhner, 2025-03-31:
Hi @mrceki ,
the  communication violation error can origin for many reasons... at the end of the day it pins down that the responses are not arriving fast enough. Did you already have a look into https://frankaemika.github.io/docs/troubleshooting.html?

Cheers,
Andreas

mrceki, 2025-04-01:
Hi @AndreasKuhner 

Yes, I did and double-checked everything. Standalone is mostly fine but I presume that when ros network is overloaded(?). Because its happening when I plot F/T data with plotjuggler or subscribe that topic with something else. Also maybe these will help to understand the issue better:

- I am using [this controller](https://github.com/mrceki/Cartesian-Impedance-Controller) on the robot.
- This also happens more frequently when the robot is in contact with the environment or some force is applied to the robot (without throwing some safety errors of course)

AndreasKuhner, 2025-04-01:
Could you maybe download the logs via Desk if you had the problem again? I would be interested to see if there is something obvious in the logs 😄

mrceki, 2025-04-02:
[logs-295341-2320154-2025-04-02T11-07-33.zip](https://github.com/user-attachments/files/19566816/logs-295341-2320154-2025-04-02T11-07-33.zip)

Here is the log file.

mrceki, 2025-04-02:
Hi @AndreasKuhner,
I ran the communication test and scored 1.0 success. Is there any other way to determine the cause of the problem?

AndreasKuhner, 2025-04-03:
Hi @mrceki ,
what image version is this log from? Is it 5.7.2 or 5.6.0?

Cheers,
Andreas

mrceki, 2025-04-03:
> Hi @mrceki ,
> what image version is this log from? Is it 5.7.2 or 5.6.0?
> 
> Cheers,
> Andreas

Hi @AndreasKuhner ,
It is 5.6.0

AndreasKuhner, 2025-04-03:
I see - so for the 5.6.0 it should be rather stable - so, make sure that on your side, there are no sudden computational spikes or network delays which could influence the latency and connection 😄

mrceki, 2025-04-03:
> I see - so for the 5.6.0 it should be rather stable - so, make sure that on your side, there are no sudden computational spikes or network delays which could influence the latency and connection 😄

I ran cyclic test to see if it is about my kernel but it is not. I ran communication test few times and get 1.0 success in all of them. That is why I am going to go crazy 😆 Only thing that fixes this is reducing the control frequency to 950 from 1000.

Could it be about libfranka or robot system version?  Currently my setup is: libfranka-v0.13.3 and robot-v5.6.0.
And could you recommend any other benchmarks to debug the issue?

AndreasKuhner, 2025-04-03:
With your libfranka + robot version, it should work quite robust. If we saw problems, it usually came from the 'outside'. E.g. heavy network traffic, heavy load on the cpu, ... 

How often do you see this error?

mrceki, 2025-04-03:
> With your libfranka + robot version, it should work quite robust. If we saw problems, it usually came from the 'outside'. E.g. heavy network traffic, heavy load on the cpu, ... 
> 
> How often do you see this error?

I am pretty sure thst cpu is not overloaded, I have a AMD 9950x.  Its probably because of network(also maybe because of ros network). 

It happens at least every minute.

AndreasKuhner, 2025-04-03:
> It happens at least every minute

Okey yes. If there would be a general problem, it would overall more present 😅 

I guess then you maybe can check if you get a better switch or try different network connections? What does your 'ping' test say, if you do it against the robot?

mrceki, 2025-04-03:
> > It happens at least every minute
> 
> Okey yes. If there would be a general problem, it would overall more present 😅
> 
> I guess then you maybe can check if you get a better switch or try different network connections? What does your 'ping' test say, if you do it against the robot?

Yes sure. My ping test resulted as 0.25ms round trip, which is nice, I think.

mrceki, 2025-04-04:
Hi @AndreasKuhner,
After tracking [this issue](https://github.com/frankaemika/franka_ros2/issues/108) in franka_ros2 repo, I downgraded libfranka version to 0.13.3 from 0.13.6, and the problem solved.

AndreasKuhner, 2025-04-04:
This seems a bit curious. Technically seen, there was a fix between those which improved the overall situation on our systems. Aka setting the thread priority to `high`. 

I would have doubts that this actually was the solution 😄

mrceki, 2025-04-04:
> This seems a bit curious. Technically seen, there was a fix between those which improved the overall situation on our systems. Aka setting the thread priority to `high`.
> 
> I would have doubts that this actually was the solution 😄

Just to confirm that, upgraded to 0.13.6 again. Here is the result:

```
[ros2_control_node-4] terminate called after throwing an instance of 'franka::ControlException'
[ros2_control_node-4]   what():  libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
```
Also, some other people have the same problem as me ([this issue](https://github.com/frankaemika/franka_ros2/issues/108)). So maybe there could be a compatibility issue.

AndreasKuhner, 2025-04-07:
> Also, some other people have the same problem as me (https://github.com/frankaemika/franka_ros2/issues/108). So maybe there could be a compatibility issue.

But this is on a different image -> 5.7.2. They report that everything works stable with a 5.6.0. for them. Having a communication violation after only a minute is also not normal 😓 . So, there might be something on your end.

mrceki, 2025-08-12:
Hi @lusu8892 , my old setup was like: robot system 5.7.2(not sure) and libfranka 0.14.1 or 0.14.2. We then downgraded the robot version to 5.6.0 + libfranka 0.13.6, but the problem persisted. Next, we tried libfranka 0.13.3, and it worked. Currently, we've upgraded the robot version to 5.8.1 and libfranka to 0.15.3, and it seems like it's working

lusu8892, 2025-08-12:
@mrceki Thanks for you quick reply

Could you help me check what is the `franka_ros2`, `franka_description` version you used for 5.6.0+0.13.3 libfranka?

Also what is the `franka_ros2`, `franka_description` version you used for 5.8.1 and libfranka to 0.15.3?

We have two setups, one plans to downgrade to 5.6.0, and one has 5.8.1 but paired with 0.15.0 libfranka. If new system works for I also want to have a try. 

Btw, I just want to share some testing result I have for now:

My setup:
PC running real-time kernal, high performance
ROS2: Humble
Robot system: 5.8.1
franka_ros2: v1.0.0
franka_description: 0.4.0
libfranka: 0.15.0

Communication_test: 100% passed

Tests have ran with ROS2 and franka_ros2:
`franka_fr3_moveit_config moveit.launch.py`, just launch this without sending any trajectory, run 10-15mins then got the error message.
`franka_bringup joint_impedance_example_controller.launch.py`, run 10-15 mins dead.

**Solely running controller example from libfranka does not cause communication_violation at all.......**
Modified joint_impedance_controller running 30mins, no error got.

This may not be conclusive, but look like either Franka_ros2 or ROS2 is not stable........

luis-camero, 2025-08-12:
@mrceki how were you able to upgrade to `5.8.1`? Through franka.world? 

I only ask because I am unable to get anything more than `5.8.0` and don't see any references to a `5.8.1` version from Franka other than on this issue thread. 

I have been having the exact same issues. On 22.04 (Humble), it seems that a single arm with all of the `cpur-freq-utils` and `realtime` permissions does the job. On 24.04 (Jazzy), I have not been as successful.

mrceki, 2025-08-12:
> @mrceki how were you able to upgrade to `5.8.1`? Through franka.world?
>
> I only ask because I am unable to get anything more than `5.8.0` and don't see any references to a `5.8.1` version from [...]

I think you can email support

luis-camero, 2025-08-12:
I have already, but thanks!

lusu8892, 2025-08-12:
> franka_ros2=0.1.15 and franka_description=0.3 or 0.4 for 5.6.0
> 
> franka_ros2=2.0.2 and franka_description=1.0.2 for 5.8.1
> 
> Maybe try upgrading libfranka to 0.15.3
> 
> Btw, I didn't test the robot that long as you did

Thanks for the information.

Just tried both setup, the newer one is still unstable, I just launch `ros2 launch franka_fr3_moveit_config moveit.launch.py` it dead after 10mins.......

Looks like the downgrade is the only way, we had 3 rounds test by running joint_impedance controller example from franka example controllers. No error met yet. 

Thanks!

mrceki, 2025-09-14:
Hi @AndreasKuhner, our system was stable as I mentioned before, but it became unstable again. The error occurs more often when you run another procces on the pc, but even a real-time kernel makes no difference.

mrceki, 2025-09-15:
Thanks for your response @AndreasKuhner ,

The other processes don't appear to be causing a significant CPU load. The purpose of a real-time kernel is to prioritise real-time processes, so it's unclear what the root of the problem is. I've also observed this issue frequently in my work with contact-rich manipulation, particularly when high external forces are involved.

Another potential factor could be the ROS communications. I'm not very familiar with its internal structure, but this might be a helpful lead for troubleshooting the issue.

davelkan, 2025-12-15:
@kmohyeldine or @AndreasKuhner could you clarify version compatibility? I've been having similar issues to those here, and have settled on the following on ros2 humble:
FR3 Firmware: 5.6.0
franka_ros2: 1.15.2
libfranka: 0.13.3
ros2_control: 2.52.2

according to [franka's compatibility table](https://frankarobotics.github.io/docs/compatibility.html) robot firmware 5.9.1 is fully compatible with franka_ros2 Humble 2.02+. We're very interested in upgrading to 5.9.1 but we had issues updating to 5.8.1 and ended up on the configuration above.

Has this also been testing using ros2_control? if so what versions?

AndreasKuhner, 2025-12-18:
Hi @davelkan ,
the latest `ros2_control` release should contain this `overrun mute` feature - which is crucial to de-activate for our robot.

Hence, if you get your hands on firmware/base image 5.9.0/5.9.1, also grap `libfranka` 0.18.0 and `franka_ros2 jazzy` 3.0.0 :)

Cheers,
Andreas

Winston-Gu, 2025-12-25:
Hi @AndreasKuhner, I am using 5.9.1 firmware, with 0.19.0 `libfranka` and `franka_ros2 jazzy` 3.0.0, but encountered the same problem