frankarobotics/libfranka #164: Incompatibility with System Version 5.7.1

pala5, 2024-10-29:
Hi all,

I attached below my system description:
- Franka Research 3 (system version 5.7.1)
- OS: Ubuntu 22.04
- ROS: ROS2 Humble distro.

The remote control PC is connected directly to Franka Shop floor Ethernet port, and the network is configured on the 172.16.0.x subnet, as it is followed is said in the [Franka Client Interface (FCI) Getting started section](https://frankaemika.github.io/docs/getting_started.html). 
I can ping the Franka without problem.

According to the [compatibility docs](https://frankaemika.github.io/docs/compatibility.html) that you have provided, I should use the libfranka version >=0.14.0. I have followed the step [3. Building and Installation](https://github.com/frankaemika/libfranka?tab=readme-ov-file#3-building-and-installation-from-source) from source instructions, but when I try to run the _comunication_test_ 
executable the following errors occurs:
`libfranka: Incompatible library version (server version: 8, library version: 9). 
Please check https://frankaemika.github.io for Panda system updates or use a different version of libfranka.

Just to be sure, I have tried with minor libfranka libraries' versions, 0.13.x, but all them are version 7. 

I am not sure what the issue is. Any ideas? There are no other libfranka installation packages, which could cause interference problems, as I am building a Docker image from source. 

Thanks!

AndreasKuhner, 2024-10-30:
Hi @pala5 ,
the libfranka library version 9 comes from the libfranka-common submodule - by any chance: After you cloned the libfranka repository and then checked out version 0.14.x, did you also make a `git submodule update`? This would pull the correct commit of libfranka-common.

Cheers,
Andreas

BarisYazici, 2024-10-31:
> According to the [compatibility docs](https://frankaemika.github.io/docs/compatibility.html) that you have provided, I should use the libfranka version >=0.14.0. I have followed the step [3. [...]

We will update the compatibility table soon. The robot image version 5.7.x is supported only with libfranka 0.14.x. 0.15.x will need another system image update possibly 5.8.x

closed 2024-10-31 by BarisYazici