frankarobotics/libfranka #167: Differential Kinematics Feature Request

jcarpinelli-bdai, 2024-11-15:
For dynamics compensation in controllers, it would be really helpful to have access to functions which compute the time derivative of the manipulator Jacobian, and potentially also the manipulator Hessian matrix.

AndreasKuhner, 2024-11-18:
Hi @jcarpinelli-bdai ,
under-the-hood we use now pinocchio in libfranka - might that one just provide all of this functionality?

Cheers,
Andreas

kmohyeldin, 2024-11-18:
Hi @jcarpinelli-bdai , you migh need to check pinocchio with [CppAD](https://github.com/coin-or/CppAD) or [CasADi](https://web.casadi.org/). in the mean while. it is not yet there in our systems

AndreasKuhner, 2025-03-05:
For any further question, please step ahead, cheers!

closed 2025-03-05 by AndreasKuhner