frankarobotics/libfranka #174: Connection is too slow - communication_constraints_violation

Minyoung1005, 2025-01-13:
Hi,

I installed franka ros2 and libfranka on an asus nuc to control franka panda research 3 robot. However, the connection seems to be slow even using real time kernel and following some advices in https://frankaemika.github.io/docs/troubleshooting.html#motion-stopped-due-to-discontinuities-or-communication-constraints-violation. We can successfully retrieve the robot's state, control the robot via gui, but can't run control scripts because they would suddenly stop due to connection problem. We've tried with or without using a network switch, but both cases raise same connection problems. We eventually want to use a gpu workstation to control the robot, so using a network switch would be necessary. Could you please help me fix this connection issue? If I can't reduce connection time, is there a way to increase the maximum time threshold?

System log is attached as a file.
[logs-295341-2320006-2025-01-13T20-36-47.zip](https://github.com/user-attachments/files/18401777/logs-295341-2320006-2025-01-13T20-36-47.zip)

Environment:
- Ubuntu 22.04 installed on asus nuc 14 pro U7 - intel core ultra U7 155H - 32GB Memory and 1TB SSD
- 5 port gigabit ethernet switch (TP-Link TL-SG105)
- Cat8 ethernet cables, lengths are in the range of 3 to 6ft

Here are some logs for running commands:

1. connection stats
```
sudo ping 172.16.0.2 -i 0.001 -D -c 10000 -s 1200

--- 172.16.0.2 ping statistics ---
10000 packets transmitted, 10000 received, 0% packet loss, time 10017ms
rtt min/avg/max/mdev = 0.202/0.830/1.076/0.185 ms

```

2. communication test occurs communication_constraints_violation
```
sudo ./examples/communication_test 172.16.0.2

WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Starting communication test.
#100 Current success rate: 0.14
#200 Current success rate: 0.20
#300 Current success rate: 0.19
#400 Current success rate: 0.16
#500 Current success rate: 0.16
#600 Current success rate: 0.17
#700 Current success rate: 0.16
#800 Current success rate: 0.15
libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
```

3. joint_impedance_example_controller.launch.py stops while running
```
ros2 launch franka_bringup joint_impedance_example_controller.launch.py arm_id:=fr3 robot_ip:=172.16.0.2

[INFO] [launch]: All log files can be found below /home/clear/.ros/log/2025-01-13-15-28-41-217496-clearnuc-9360
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [joint_state_publisher-1]: process started with pid [9361]
[INFO] [robot_state_publisher-2]: process started with pid [9363]
[INFO] [ros2_control_node-3]: process started with pid [9365]
[INFO] [spawner-4]: process started with pid [9367]
[INFO] [spawner-5]: process started with pid [9369]
[INFO] [franka_gripper_node-6]: process started with pid [9371]
[INFO] [spawner-7]: process started with pid [9373]
[ros2_control_node-3] [WARN] [1736800121.520665321] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1736800121.520890768] [resource_manager]: Loading hardware 'FrankaHardwareInterface' 
[franka_gripper_node-6] [INFO] [1736800121.530114112] [fr3_gripper]: Trying to establish a connection with the gripper
[franka_gripper_node-6] [INFO] [1736800121.531454168] [fr3_gripper]: Connected to gripper
[ros2_control_node-3] [INFO] [1736800121.540168305] [resource_manager]: Initialize hardware 'FrankaHardwareInterface' 
[ros2_control_node-3] [INFO] [1736800121.540212987] [FrankaHardwareInterface]: Connecting to robot at "172.16.0.2" ...
[robot_state_publisher-2] [INFO] [1736800121.544211110] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1736800121.544305369] [robot_state_publisher]: got segment fr3_hand
[robot_state_publisher-2] [INFO] [1736800121.544310114] [robot_state_publisher]: got segment fr3_hand_tcp
[robot_state_publisher-2] [INFO] [1736800121.544312791] [robot_state_publisher]: got segment fr3_leftfinger
[robot_state_publisher-2] [INFO] [1736800121.544315199] [robot_state_publisher]: got segment fr3_link0
[robot_state_publisher-2] [INFO] [1736800121.544317537] [robot_state_publisher]: got segment fr3_link1
[... 277 lines cut ...]
analyzing CPU 20:
  driver: intel_pstate
  CPUs which run at the same hardware frequency: 20
  CPUs which need to have their frequency coordinated by software: 20
  maximum transition latency: 4294.55 ms.
  hardware limits: 400 MHz - 2.50 GHz
  available cpufreq governors: performance, powersave
  current policy: frequency should be within 400 MHz and 2.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency is 1.60 GHz.
analyzing CPU 21:
  driver: intel_pstate
  CPUs which run at the same hardware frequency: 21
  CPUs which need to have their frequency coordinated by software: 21
  maximum transition latency: 4294.55 ms.
  hardware limits: 400 MHz - 2.50 GHz
  available cpufreq governors: performance, powersave
  current policy: frequency should be within 400 MHz and 2.50 GHz.
                  The governor "performance" may decide which speed to use
                  within this range.
  current CPU frequency is 1.60 GHz.
```

Thanks

AndreasKuhner, 2025-01-14:
Hi @Minyoung1005 ,
I guess the problem is the round-trip time:

```
--- 172.16.0.2 ping statistics ---
10000 packets transmitted, 10000 received, 0% packet loss, time 10017ms
rtt min/avg/max/mdev = 0.202/0.830/1.076/0.185 ms
```

We need a round-time time of at max 0.3 to 0.4ms -> you are in mean at 0.8ms. So, I think you need to get another network card or network connection from your PC. Maybe you could try to start with a 'simple' notebook running an RT to get it to run but you need to get this round-trip time down by a lot.

Cheers,
Andreas

Winston-Gu, 2025-12-25:
Hi @Minyoung1005, I think I have met exactly the same issue these days. Could you please share some insights on how you eventually fixed this issue?