frankarobotics/libfranka #177: Inquiry About Real-Time Gripper Control for Teleoperation

yiyikkk, 2025-02-03:
I am currently working on a project that involves teleoperating a Franka Panda FR3 robot accurately using a leader arm, and I have some questions regarding the real-time control capabilities of the gripper.
 
In our setup, we are using the move and grasp commands to control the gripper. However, we encountered a few limitations:
 
- The move command allows us to set the gripper width, but it is not practical for teleoperation since the object width is unknown in advance, leading to potential hangs if the width is smaller than the object. Also no force applied by using this command.
 
- The grasp command works by closing the gripper until a contact is detected, but it is not designed for real-time teleoperation either. It executes a single grasping action rather than dynamically following the position of the teleoperated leader arm’s gripper.
 
- Both move and grasp commands take more than one second to execute, which makes it impossible to achieve real-time teleoperation.
 
**Our goal is to achieve accurate real-time control where the Franka gripper dynamically follow the opening and closing of the leader arm’s gripper.** Specifically:
 
- If the leader arm’s gripper closes to 20%, Franka’s gripper should also close to 20%, regardless of object presence.
 
- If the leader arm’s gripper closes beyond the object width, Franka should grasp the object and maintain the grip as long as the leader arm’s gripper remains more closed than the object width.
 
- Ideally, the tighter Gello’s grip, the greater the force applied by Franka.
 
Does anyone know if it is possible to achieve this level of real-time control with the current Franka Hand? If not, are there any alternative setups or configurations you would recommend to support such functionality?
 
I appreciate any suggestions and assistance.

danielsanjosepro, 2025-02-17:
I am also interested in knowing if this is even possible. I was not able to find any ressources online

AndreasKuhner, 2025-02-17:
Hi @yiyikkk ,
at the moment only the rather slow commanding style for the gripper is possible. We are aware of the issue but due the current technical limitations in the backend, there is also now quick fix for that. Rather, we need a new control scheme to make it possible. 

But, we are on it! Release... well 😅 

Cheers,
Andreas

yiyikkk, 2025-02-18:
> Hi [@yiyikkk](https://github.com/yiyikkk) , at the moment only the rather slow commanding style for the gripper is possible. We are aware of the issue but due the current technical limitations in [...]

Thank you for confirming my understanding! Good to know it, and look forward to the updates in the future 😄

closed 2025-02-18 by yiyikkk