frankarobotics/libfranka #180: SetCartesianImpedance has no effect

pauljansonnie, 2025-02-17:
Hi,

I noticed that the values specified in `SetCartesianImpedance` are ignored (i.e. no noticeable effect on stiffness of robot). I am running the latest libfranka from master (i.e. roughly 0.15.0 I supposed) that I build from source and a FR3 with firmware 5.7.2.

How to replicate (full code below): Modify  the example `generate_cartesian_pose_motion.cpp` where:
- call `robot.control()` with controller mode `kCartesianImpedance`,
- set the cartesian impedance using `SetCartesianImpedance`, 
- set more compliant collision behavior using `robot.setCollisionBehavior` (optional)

When running the example, whatever the value of the cartesian stiffness I specify, the robot is always as stiff. When doing a similar modification but with for joint impedance (replace `kCartesianImpedance` with `kJointImpedance`, replace `SetCartesianImpedance` with `SetJointImpedance`), the robot stiffness changes based on the set values. It suggests that the problem only occurs in Cartesian mode.

Am I doing something wrong? Is there a way to check the set stiffness? Please let me know.

```cpp
// Copyright (c) 2023 Franka Robotics GmbH
// Use of this source code is governed by the Apache-2.0 license, see LICENSE
#include <cmath>
#include <iostream>

#include <franka/exception.h>
#include <franka/robot.h>

#include "examples_common.h"

/**
 * @example generate_cartesian_pose_motion.cpp
 * An example showing how to generate a Cartesian motion.
 *
 * @warning Before executing this example, make sure there is enough space in front of the robot.
 */

int main(int argc, char** argv) {
  if (argc != 2) {
    std::cerr << "Usage: " << argv[0] << " <robot-hostname>" << std::endl;
    return -1;
  }
  try {
    franka::Robot robot(argv[1]);
    setDefaultBehavior(robot);
    robot.SetCartesianImpedance({{100.0, 100.0, 100.0, 10.0, 10.0, 10.0}});

    // First move the robot to a suitable joint configuration
    std::array<double, 7> q_goal = {{0, -M_PI_4, 0, -3 * M_PI_4, 0, M_PI_2, M_PI_4}};
    MotionGenerator motion_generator(0.5, q_goal);
    std::cout << "WARNING: This example will move the robot! "
              << "Please make sure to have the user stop button at hand!" << std::endl
              << "Press Enter to continue..." << std::endl;
    std::cin.ignore();
    robot.control(motion_generator, franka::ControllerMode::kCartesianImpedance);
    std::cout << "Finished moving to initial joint configuration." << std::endl;

    // Set additional parameters always before the control loop, NEVER in the control loop!
    // Set collision behavior.
    robot.setCollisionBehavior({{100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0}},
                               {{100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0}},
                               {{100.0, 100.0, 100.0, 100.0, 100.0, 100.0}},
                               {{100.0, 100.0, 100.0, 100.0, 100.0, 100.0}});

    std::array<double, 16> initial_pose;
    double time = 0.0;
    robot.control([&time, &initial_pose](const franka::RobotState& robot_state,
                                         franka::Duration period) -> franka::CartesianPose {
      time += period.toSec();

      if (time == 0.0) {
        initial_pose = robot_state.O_T_EE;
      }

      constexpr double kRadius = 0.3;
      double angle = M_PI / 4 * (1 - std::cos(M_PI / 5.0 * time));
      double delta_x = kRadius * std::sin(angle);
      double delta_z = kRadius * (std::cos(angle) - 1);

      std::array<double, 16> new_pose = initial_pose;
      new_pose[12] += delta_x;
      new_pose[14] += delta_z;

      if (time >= 10.0) {
        std::cout << std::endl << "Finished motion, shutting down example" << std::endl;
        return franka::MotionFinished(new_pose);
      }
      return new_pose;
    },
    franka::ControllerMode::kCartesianImpedance
  );
  } catch (const franka::Exception& e) {
    std::cout << e.what() << std::endl;
    return -1;
  }

  return 0;
}
```

AndreasKuhner, 2025-02-17:
Hi @pauljansonnie ,
theoretically, it should work. Can you try to set the Cartesian impedance after the first `.control` and before the second?

It might be that the finish of the first control might override the initial set value (and brings the robot to a default state).

Cheers,
Andreas

pauljansonnie, 2025-02-17:
Hi @AndreasKuhner,

The behavior remains the same (very stiff arm). As I know you are also checking franka_ros2 issues, I am letting you know that this issue is also reflected when using the service set_cartesian_stiffness in franka_ros2 (i.e. even if request from client is successful, arm remains just as stiff). Would it be possible to check if you can replicate the issue?

pauljansonnie, 2025-02-17:
> In franka_ros2 I know that it can't work - there we always use joint_impedance control as control mode 😅 .

I forgot to mention that I manually edited the code for the cartesian impedance control mode to be used, as suggested here in https://github.com/frankaemika/franka_ros2/issues/53#issuecomment-2034053397.

pauljansonnie, 2025-03-10:
@AndreasKuhner Any update on this? If no upcoming fix, where can I find guidelines to downgrade the firmware of the robot?

kmohyeldine, 2025-09-25:
@mgstanley it fixed, and will be in the next system image release in mid october aprox.

mgstanley, 2025-09-25:
Thank you so much for the good news!