frankarobotics/libfranka #182: Forgot password for login the franka emika panda Desk through Browser

yijionglin, 2025-02-21:
Hi there,

We forgot the password to log in to the Desk through a new computer. However, we still can access the Desk with an old computer which had logged in before and it doesn't need the password.

Is there any way to reset the password? We need to access the robot with the new PC...

All the best,

AndreasKuhner, 2025-03-04:
Hi @yijionglin ,
I think you can try to factory reset it which allows you to set a new user etc.

Cheers,
Andreas

yijionglin, 2025-03-04:
Instead of performing a factory reset (which could potentially update the firmware and cause incompatibility with existing code), I solved this problem by adding new accounts through the old PC—without needing to know the password for the old account!

Bravo!

closed 2025-03-04 by yijionglin

AndreasKuhner, 2025-03-04:
This works as well, good job!