frankarobotics/libfranka #190: 🔄 We’re Moving to frankarobotics — June 15 #140

kmohyeldine, 2025-06-05:
Important Notice: Repository Migration

This repository—and all others under the frankaemika GitHub organization—will be migrated to our new home at:

🔗 https://github.com/frankarobotics

This change reflects our transformation into Franka Robotics, with a renewed commitment to developer experience, openness, and innovation.

What You Need to Know:
This repository will no longer be maintained here after June 15, 2025

Web links to repositories will redirect automatically, but git clone via old URLs will no longer work

You may continue pushing to old remotes for a limited time

Please update your remotes, scripts, bookmarks, and documentation

Thank you for building with us!
We look forward to continuing the journey at [frankarobotics](https://github.com/frankarobotics) 🚀

Haran100, 2025-06-20:
Is there an ETA on when the documentation will be back up? 
https://frankaemika.github.io/docs/libfranka.html
Thanks

AndreasKuhner, 2025-06-24:
Hi @Haran100 ,
it is up again - but with different URL: https://frankarobotics.github.io/docs/libfranka.html

Cheers,
Andreas

ggb367, 2025-07-09:
Hi @AndreasKuhner, do you know when we can expect to see the documenation at [https://frankaemika.github.io/libfranka/0.15.0/](https://frankaemika.github.io/libfranka/0.15.0/) to be back online? Thanks!

closed 2025-07-10 by kmohyeldine