frankarobotics/libfranka #201: Unexpected behavior with franka::Robot::setLoad

hesic73, 2025-09-04:
Hello, I have a question about [franka::Robot::setLoad](https://docs.ros.org/en/kinetic/api/libfranka/html/classfranka_1_1Robot.html#afcb708df10f24563dbcf7d5b907b4a15).

**Environment**:

- libfranka 0.15.3
- Franka Research 3 Control: 5.7.2

I'm using the following script to test setting and reading load parameters:

```C++
#include <iostream>
#include <array>
#include <iomanip>
#include <chrono>
#include <thread>

#include <franka/exception.h>
#include <franka/robot.h>

/**
 * @brief A helper function to print the contents of a std::array.
 */
template<typename T, size_t N>
void print_array(std::ostream& os, const std::array<T, N>& arr) {
    os << "[";
    for (size_t i = 0; i < N; ++i) {
        os << arr[i] << (i == N - 1 ? "" : ", ");
    }
    os << "]";
}

/**
 * @brief Prints the configured external load parameters from a RobotState.
 * @param robot_state The robot state to read the parameters from.
 */
void printLoadParameters(const franka::RobotState& robot_state) {
    std::cout << std::fixed << std::setprecision(4);
    std::cout << "Mass (m_load):         " << robot_state.m_load << " kg" << std::endl;
    std::cout << "Center of Mass (F_x_Cload): ";
    print_array(std::cout, robot_state.F_x_Cload);
    std::cout << " m" << std::endl;
    std::cout << "Inertia Tensor (I_load): " << std::endl;
    // Note: libfranka's inertia matrix is column-major. We print it in row-major format for readability.
    std::cout << "  [" << std::setw(8) << robot_state.I_load[0] << ", " << std::setw(8) << robot_state.I_load[3] << ", " << std::setw(8) << robot_state.I_load[6] << "]" << std::endl;
    std::cout << "  [" << std::setw(8) << robot_state.I_load[1] << ", " << std::setw(8) << robot_state.I_load[4] << ", " << std::setw(8) << robot_state.I_load[7] << "]" << std::endl;
    std::cout << "  [" << std::setw(8) << robot_state.I_load[2] << ", " << std::setw(8) << robot_state.I_load[5] << ", " << std::setw(8) << robot_state.I_load[8] << "]" << std::endl;
}

/**
 * @brief Main function to connect to a Franka robot, read its state once,
 * and print the configured external load parameters.
 */
int main(int argc, char** argv) {
    if (argc != 2) {
        std::cerr << "Usage: " << argv[0] << " <robot-hostname>" << std::endl;
        std::cerr << "Example: " << argv[0] << " 172.16.0.2" << std::endl;
        return -1;
    }

    try {
        franka::Robot robot(argv[1]);
        std::cout << "Successfully connected to robot at " << argv[1] << std::endl;

        std::cout << "\n--- Reading Initial Load ---" << std::endl;
        printLoadParameters(robot.readOnce());

        std::cout << "\n--- Setting New Load ---" << std::endl;
        double new_mass = 0.5; // kg
        std::array<double, 3> new_F_x_Cload = {0.0, 0.0, 0.08}; // m
        std::array<double, 9> new_inertia = {0.001, 0, 0, 0, 0.001, 0, 0, 0, 0.001}; // A simple diagonal inertia matrix
        robot.setLoad(new_mass, new_F_x_Cload, new_inertia);
        std::cout << "Set command sent. Waiting for 2 seconds..." << std::endl;
        std::this_thread::sleep_for(std::chrono::seconds(2));

        std::cout << "\n--- Reading Load After Setting ---" << std::endl;
        printLoadParameters(robot.readOnce());

        std::cout << "\n--- Resetting Load to Zero ---" << std::endl;
        robot.setLoad(0.0, {0.0, 0.0, 0.0}, {0, 0, 0, 0, 0, 0, 0, 0, 0});
        std::cout << "Reset command sent. Waiting for 2 seconds..." << std::endl;
        std::this_thread::sleep_for(std::chrono::seconds(2));
        
        std::cout << "\n--- Reading Load After Resetting ---" << std::endl;
        printLoadParameters(robot.readOnce());

    } catch (const franka::Exception& e) {
        std::cerr << "Franka Exception: " << e.what() << std::endl;
        return -1;
    }

    return 0;
}
```

**Expected behavior**: The load parameters should be set exactly as specified and read back with the same values.

```
Successfully connected to robot at 169.254.67.230

--- Reading Initial Load ---
Mass (m_load):         0.0000 kg
Center of Mass (F_x_Cload): [0.0000, 0.0000, 0.0000] m
Inertia Tensor (I_load): 
  [  0.0000,   0.0000,   0.0000]
  [  0.0000,   0.0000,   0.0000]
  [  0.0000,   0.0000,   0.0000]

--- Setting New Load ---
Set command sent. Waiting for 2 seconds...

--- Reading Load After Setting ---
Mass (m_load):         0.5000 kg
Center of Mass (F_x_Cload): [0.0000, 0.0000, 0.0800] m
Inertia Tensor (I_load): 
  [  0.0010,   0.0000,   0.0000]
  [  0.0000,   0.0010,   0.0000]
  [  0.0000,   0.0000,   0.0010]

--- Resetting Load to Zero ---
Reset command sent. Waiting for 2 seconds...

--- Reading Load After Resetting ---
Mass (m_load):         0.0000 kg
Center of Mass (F_x_Cload): [0.0000, 0.0000, 0.0000] m
Inertia Tensor (I_load): 
  [  0.0000,   0.0000,   0.0000]
  [  0.0000,   0.0000,   0.0000]
  [  0.0000,   0.0000,   0.0000]

```

**Actual behavior**: Most of the time, the parameters are not set to the exact values I specify

```
Successfully connected to robot at 169.254.67.230

--- Reading Initial Load ---
Mass (m_load):         0.0000 kg
Center of Mass (F_x_Cload): [0.0000, 0.0000, 0.0000] m
Inertia Tensor (I_load): 
  [  0.0000,   0.0000,   0.0000]
  [  0.0000,   0.0000,   0.0000]
  [  0.0000,   0.0000,   0.0000]

--- Setting New Load ---
Set command sent. Waiting for 2 seconds...

--- Reading Load After Setting ---
Mass (m_load):         0.4094 kg
Center of Mass (F_x_Cload): [0.0000, 0.0000, 0.0655] m
Inertia Tensor (I_load): 
  [  0.0008,   0.0000,   0.0000]
  [  0.0000,   0.0008,   0.0000]
  [  0.0000,   0.0000,   0.0008]

--- Resetting Load to Zero ---
Reset command sent. Waiting for 2 seconds...

--- Reading Load After Resetting ---
Mass (m_load):         0.0906 kg
Center of Mass (F_x_Cload): [0.0000, 0.0000, 0.0145] m
Inertia Tensor (I_load): 
  [  0.0002,   0.0000,   0.0000]
  [  0.0000,   0.0002,   0.0000]
  [  0.0000,   0.0000,   0.0002]

```

Could anyone provide any insight into this? Thanks!

kmohyeldine, 2025-09-09:
Thanks @hesic73 we are on that1

BarisYazici, 2025-09-29:
@hesic73 is the issue happening sporadically or every time? I am trying to reproduce the issue in image 5.8.1 with your example code but I never got different values than I set. Do you set anything else before you execute your code? 

Also try to upgrade to the 5.8.1 there are some performance upgrades for FCI users. And please try again if the issue still persists.

hesic73, 2025-10-05:
Thanks for the follow-up. Unfortunately, I don't use franka these days, so I can't reproduce the issue or test with the updated software. I was just curious about this API's behavior, particularly for scenarios like moving heavy objects with the gripper, as it seems relevant for proper control. I'll close this issue.

closed 2025-10-05 by hesic73